harbaum/galagino #1: Did you ever see these failures?

danjulio, 2023-01-26:
Hi Till,

First, thanks for this wonderful project.  I have ported it to my board with a WROVER and 320x480 pixel ILI9488 display.  It works but with a few failures I am wondering if you can shed some light on (for example you saw this during your development).

1. I get no sound.  Nothing coming out of the DAC pins (25 & 26) even when viewed with an oscilloscope.
2. The emulator (or at least parts of it) seem to freeze on occasion.
  - If I press COIN after it has displayed the attract screen for a while movement of the sprites stops and I can add no more coins but the starry background keeps moving.  However if I hold COIN down through the startup then I immediate get a coin and can others and can start the play.
  - After a game it freezes after displaying the statistics.  The statistics go away and the the entire game freezes (starry background freezes too).
  
I am compiling on Arduino 1.8.19 (Mac) with ESP Arduino package 2.0.6.  I downloaded my ROMs from https://wowroms.com/en/roms/mame-0.139u1/download-galaga-namco-rev.-b/3699.html.  All your python scripts seemed to work just fine on the Mac.  Board is https://github.com/danjulio/gCore.
  
  Changes I made include
  1. I display a background image instead of blanking the LCD as you can see to make use of the extra LCD real-estate (seems to work fine).
  2. I used different IO pins for the buttons (33, 34, 35, 36, 39) and used external 10k-ohm pull-ups (seems to work fine).
  3. I used the TFT_eSPI library to drive the LCD (80 MHz VSPI native pins).  Seems to work fine and I did  a little playing with your code to try out 60 Hz.  Seems to work - starry background scrolls a lot faster - but didn't get an actual FPS yet.
  
I realize it is really hard to debug someone else's setup but just curious if you had seen any of these issues or had a pointer where I might go debugging.
  
 Thanks, Dan

[image]

harbaum, 2023-01-26:
Great that you are having fun with it.

And no, I have not seen any of these issues. Actually my kids just yesterday stress tested the current version for over an hour with no complaints at all. 

I would suggest that you go step-by-step from my hw setup to yours to see where things start to break. But I assume you already had your setup lying around. Interestingly your button layout matches the one required for Galaga. What was you board designed for?

One thing that comes to mind the the WROVER. Do you select a WROVER board in the Arduino IDE? If yes, the code may actually use the external RAM instead of the internal one. As the external RAM is much slower this may be related to your problems. Try to select a WROOM based board which should make the code just ignore the extra RAM of the WROVER.

I have done some major modifications in the meantime and in a few days I'll release a new version with options to include *PACMAN* and *Donkey Kong* as together with a game select screen. This is optional. If you prefer a bare Galaga you can still set it up as a single machine.

danjulio, 2023-01-26:
Thank you for your quick response!  I got it running.  And as we all know that kids are the perfect stress test for anything, I'm sure it's golden now :-)

1. You were right, it has to be compiled for WROOM.  Something must have been getting malloc'd in PSRAM.
2. I replaced your call to i2s_set_pin() with i2s_set_dac_mode(I2S_DAC_CHANNEL_BOTH_EN) to get audio working.  I also found that i2s_driver_install() starts the driver as well so technically the i2s_start() call is not needed.  The only thing I can think is that we're using different versions of the ESP32 Arduino package and Espressif changed something in between.

Thank you again.  I look forward to your updates.  I spent some time researching how the galaga console worked and how you implemented the emulation.  Don't understand it all but I think it is very cool.

Regards, Dan

harbaum, 2023-01-30:
> The only thing I can think is that we're using different versions of the ESP32 Arduino package and Espressif changed something in between.

I am running ESP32 board package 1.0.6 on a windows box as well as a linux box and it compiles with that.

danjulio, 2023-01-30:
It's all good!  I got it running.  And I see you're making progress with the other games.  Very cool.  Thanks again for helping me.

closed 2023-01-30 by danjulio

harbaum, 2023-01-31:
I changed to version 2.0.6 as well and lost the audio. Just replacing  i2s_set_pin() with i2s_set_dac_mode(I2S_DAC_CHANNEL_BOTH_EN) does not give me audio back.

harbaum, 2023-01-31:
It seems the magic is not calling ```i2s_set_dac_mode(I2S_DAC_CHANNEL_BOTH_EN)``` but removing the setting of ```.communication_format = (i2s_comm_format_t)I2S_COMM_FORMAT_I2S_MSB``` from ```i2s_config```.

danjulio, 2023-01-31:
Interesting.  And I have to apologize.  In fact I changed that too but forgot to mention it.  I saw there was an incorrect cast after looking at the driver code.  Here's my setup that's working.  I set ```.communication_format = I2S_COMM_FORMAT_STAND_MSB,```  without the cast, but as you found out, it isn't necessary.

``` 
// init audio
  // 24 kHz @ 16 bit = 48000 bytes/sec = 800 bytes per 60hz game frame =
  // 1600 bytes per 30hz screen update = ~177 bytes every four tile rows  
  static const i2s_config_t i2s_config = {
    .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_TX | I2S_MODE_DAC_BUILT_IN),
    .sample_rate = 24000,
    .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,
#ifdef SND_DIFF
    .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
#else
    .channel_format = I2S_CHANNEL_FMT_ONLY_LEFT,
#endif
//    .communication_format = (i2s_comm_format_t)I2S_COMM_FORMAT_I2S_MSB,
    .communication_format = I2S_COMM_FORMAT_STAND_MSB,
    .intr_alloc_flags = 0,
    .dma_buf_count = 4,
    .dma_buf_len = 64,   // 64 samples
    .use_apll = false
  };

  snd_prepare();
  snd_render_buffer();
  i2s_driver_install(I2S_NUM_0, &i2s_config, 0, NULL);
  // use dac on default gpio25/26
//  i2s_set_pin(I2S_NUM_0, NULL);
  i2s_set_dac_mode(I2S_DAC_CHANNEL_BOTH_EN);
//  i2s_start(I2S_NUM_0);
```

harbaum, 2023-01-31:
I just checked in an updated version after I updated to 2.0.6 as well. This should now work for you as well.

harbaum, 2023-02-04:
FYI: The latest code runs the video at full 60Hz if SPI clock is configured to 80Mhz. This might also work with your display.

danjulio, 2023-02-04:
Thanks Till!  I gotta make a new button board now :-)