harbaum/galagino #2: 80Hz SPI on ILI9341

gingerbeardman, 2023-01-26:
I read 80Hz SPI is possible on ILI9341:

https://www.esp32.com/viewtopic.php?t=6627

harbaum, 2023-01-27:
Some display do. But in general they don't. But there's imho no visible benefit, anyways. The video is very fluid at 30Hz.

gingerbeardman, 2023-01-27:
Appreciate your reply.

My goal would be the same refresh rate as the original arcade hardware. I'll keep looking!

harbaum, 2023-01-27:
The st7789 displays I have tested work fine at 80Mhz and one of my ili9143 also copes with that. 

Just give it a try. You'll have to do some minor adjustments in the rest of the code as well to e.g. make sure that the code execution will still get 60 VBL interrupts per second to make the game run at the correct speed.

Also be aware that being able to write data at the full rate into the display controller doesn't guarantee that the display is actually refreshed at that rate. You'll probably need to study the data sheets of the display controllers in more detail to verify that these refresh those small screens at the desired rate.

However, why don't you just give it a try?

gingerbeardman, 2023-01-27:
Thanks again.

Oh, also, I added this to Hacker News and there's some discussion there: https://news.ycombinator.com/item?id=34539811

harbaum, 2023-01-27:
Regarding that discussion: Doing partial screen updates for Galaga is difficult as the starfield background basically affects the entire screen the whole time. That combined with the three z80 utilizes the ESP32 to ~90%. At least with my C code. Using assembly would sure relax things.

Pacman and Donkey Kong are a different thing. They are both single Z80 machines and the screen content is rather static. So yes, these run easily on the ESP32 and may even be possible with the ESP8266.

closed 2023-02-04 by harbaum