harbaum/galagino #5: I2S_MODE_DAC_BUILT_IN was not declared in void audio_init

famiclone6502, 2023-01-30:
I've been trying/failing to compile/verify the .ino while I am waiting for my ESP32 to arrive. 

Right now I've got an error in void audio_init() that I2S_MODE_DAC_BUILT_IN was not declared. Do I need a specific version of the ESP32 libraries? Can you provide which versions you are using?

On another note, I saw that the readme is now missing the section for using the other .py scripts, which I was able to find to get past the tileaddr.h error. While it did not mention the game logo python script, I was able to figure out the usage and run that command too. 

Thank you and I look forward to building this project!

harbaum, 2023-01-31:
Issues with I2S have also been reported in #1. I'll check that. I am using ESP32 board support 1.0.6 which comes from https://dl.espressif.com/dl/package_esp32_index.json . Where does 2.06 come from?

The main [README.md](README.md) now has links to the romconv and samples directory. These now have their own [romconv/README.md](romconv/README.md) [samples/README.md](samples/README.md). But I have added another sentence to the main README which points to these. This should be more obvious now.

harbaum, 2023-01-31:
It seems 2.0.6 comes from [here](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html). I'll have a look at that.

harbaum, 2023-01-31:
You need the latest 2.0.6 ESP32 board support as described e.g. here: https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html

famiclone6502, 2023-01-31:
I was able to downgrade to 1.0.6 and successfully verify the build. I don't have the board yet so I can't check if everything else went correctly. Thanks for updating the readme and letting me know which version to try. I recommend adding which libraries have been tested with the project to the readme, as even future updates could break compatibility further. Are you planning on updating it to work with the current latest version?

harbaum, 2023-01-31:
I just updated it to work with 2.0.6 which is the latest. I actually haven't tested if this still works with 1.0.6 but I don't expect it to work. So you might have to update to 2.0.6

famiclone6502, 2023-01-31:
Thank you for clarifying, I was able to start from scratch and successfully compile using the updated readme instructions and latest versions of your project with the 2.0.6 esp32 library. 

I forgot to mention another dependency you may want to add to the readme, which is using "pip3 install imageio" before converting the logos with Python.

Thanks again!

famiclone6502, 2023-02-02:
Thank you for updating the documentation. I was able to complete my project. (The original screen was broken and the board was corroded).
[image]

The only other documentation issue I had was figuring out the buttons, which is listed in the code but unlabeled on the breadboard schematic. I did end up flipping the screen, so that feature was a perfect inclusion. 

But you can close this issue, now. Thank you for making this! It was a very fun and easy project to build! I look forward to any future updates!

closed 2023-02-02 by harbaum