harbaum/galagino #8: Combine coin and start button function

harbaum, 2023-02-01:
On a real arcade machine inserting a coin and starting a game are two different things. On the emulator inserting a virtual coin and starting always happen in a sequence.

Implement a automatic that is triggered by the start button and automatically inserts a coin before. The coin button can then be totally omitted.

famiclone6502, 2023-02-03:
I agree, but also would also like to combine the action button, too. For my project, I used a second joystick for these buttons with left being coin, right as start, and down is action (jump/fire).  It's a little awkward, so I would prefer to remove this joystick and just replace it with a single button if it's feasible.

harbaum, 2023-02-04:
How would this work? Combining coin and start works as the user never uses these buttons for other purposes. But how would you wire the fire button into this? Mapping one single button onto "fire" and "coin" and "start" would cause you to constantly insert coins while playing and hitting fire.

famiclone6502, 2023-02-04:
Sorry, I meant more of a freeplay mode where you only need one button to start the game, and use the same button for action.

harbaum, 2023-02-04:
Freeplay can already be enabled in the config. That reminds me that I have only implemented that for galaga and Pacman. Does the Donkey Kong allow for Freeplay as well? I need to check that.

However, you can enable this and this should remove the need for the coin button. Then you can simply wire fire and start to one single button. That should already allow you to do a single button setup.

closed 2023-02-05 by harbaum

famiclone6502, 2023-02-06:
I do think this is an interesting way to add freeplay to combine the buttons, but like you said it still adds coins while playing. This is more scalable than finding patches for each individual game (if you plan on adding more), but if any credits remain after a game, the attract mode won't work. 

Patches are out there to give these games better freeplay without disabling the attract mode. I tried one for Pacman (from Cam's Pacman page) that doesn't add credits if you press a button (and says "FREEPLAY"), but randomly it seems to get halfway stuck on the attract screen when starting. Another good page for patches covering the games supported so far would be the ones on SouzaOnline, like the 4-in-1 that lets you choose fast/slow along with adding Ms Pacman. But, I wasn't sure how to incorporate these since they didn't have normal ROM names/sizes.  There's also a freeplay patch for Donkey Kong there, too. 

I think most true freeplay patches would allow a single button for action and starting without hearing credits being added, but it's understandable if this does not align with your goals for this project.