harbaum/galagino #13: missing tileaddr.h: No such file

C64fan, 2023-03-25:
Hi, 
great project, respect.
I cannot find tileaddr.h , where to get this file ?

harbaum, 2023-03-25:
You generate it together with many other files using the python scripts in  the romconv directory. 

There's a readme that should give you all the details.

C64fan, 2023-03-25:
My bad,
Ordered hardware and will build this project.
Indeed very detailed manual, very appreciated.

closed 2023-03-25 by C64fan

harbaum, 2023-03-25:
This is still a work in progress and the next will be a laser cut plywood case.

And the instructions will get a major update then to include the mechanical, electrical and programming related tasks in one consistent document.

Do you plan to put everthing into a case?