harbaum/galagino #14: galaga_starseed.h

C64fan, 2023-03-25:
Hi,
progressing on the project.
All the python scripts succeeded on the roms but with compile of the project the galaga_starseed.h is throwing struct errors.
I read the how-to, do i fail with this file ?
Thanks in advance

harbaum, 2023-03-25:
What are those errors?

C64fan, 2023-03-25:
Hi Till,  In file galagino\galagino.ino:41:galagino\galaga_starseed.h:1:1: error: expected primary-expression before 'struct' struct galaga_star { unsigned char x, y; unsigned short col; }; ^~~~~~galagino\galaga_starseed.h:1:1: error: expected '}' before 'struct'In file galagino\galagino.ino:40:galagino\galaga_sample_boom.h:1:44: note: to match this '{' const unsigned char galaga_sample_boom[] = {                                            ^In file galagino\galagino.ino:41:galagino\galaga_starseed.h:2:26: error: elements of array 'const galaga_star galaga_star_set [][63]' have incomplete type const struct galaga_star galaga_star_set[][63] = {                          ^~~~~~~~~~~~~~~galagino\galaga_starseed.h:2:26: error: storage size of 'galaga_star_set' isn't knowngalagino\galagino.ino: In function 'void render_stars_set(short int, const galaga_star*)':galagino\galagino.ino:218:38: error: invalid use of incomplete type 'struct galaga_star'     const struct galaga_star *s = set+star_cntr;                                   ~~~^~~~~~~~~~In file galagino\galagino.ino:41:galagino\galaga_starseed.h:2:14: note: forward declaration of 'struct galaga_star' const struct galaga_star galaga_star_set[][63] = {              ^~~~~~~~~~~\galagino.ino:220:32: error: invalid use of incomplete type 'const struct galaga_star'     unsigned short x = (244 - s->x) & 0xff;                                ^~In file galagino\galagino.ino:41:galagino\galaga_starseed.h:2:14: note: forward declaration of 'struct galaga_star' const struct galaga_star galaga_star_set[][63] = {              ^~~~~~~~~~~galagino\galagino.ino:221:27: error: invalid use of incomplete type 'const struct galaga_star'     unsigned short y = ((s->y + stars_scroll_y) & 0xff) + 16 - row * 8;                           ^~In file galagino\galagino.ino:41:galagino\galaga_starseed.h:2:14: note: forward declaration of 'struct galaga_star' const struct galaga_star galaga_star_set[][63] = {              ^~~~~~~~~~~galagino\galagino.ino:224:34: error: invalid use of incomplete type 'const struct galaga_star'       frame_buffer[224*y + x] = s->col;                                  ^~In file galagino\galagino.ino:41:galagino\galaga_starseed.h:2:14: note: forward declaration of 'struct galaga_star' const struct galaga_star galaga_star_set[][63] = {              ^~~~~~~~~~~ Compilation error: expected primary-expression before 'struct' A snip of the file  Thank you for your help and time, really nice project. GreetsC64fan Van: Till HarbaumVerzonden: zaterdag 25 maart 2023 21:21Aan: harbaum/galaginoCC: C64fan; AuthorOnderwerp: Re: [harbaum/galagino] galaga_starseed.h (Issue #14) What are those errors?

C64fan, 2023-03-26:
Hi,
I will close this question to not "spoil" this great project, really great effort.
Question is still open, i will wait for your answer or tip to solve the scruct error.
It can be related to Arduino Ide, or library or used setting only galaga in config.
Thank you Till for this nice project.
C64fan

closed 2023-03-26 by C64fan

harbaum, 2023-03-26:
I just did a build from scratch and it worked as it should. So there's something different in your setup.

My assumption is that one of the files was not created properly. It seems the first line if `galaga_starseed.h` is not parsed properly and thus the following second line in `galaga_starseed` fails. 

My guess is that the file `galaga_sample_boom.h` is broken as it's included right before `galaga_starseed.h` . Can you have a look at that file? It shouls start with `const unsigned char galaga_sample_boom[] = {` and end with `};` in the last line.

BTW: Where and how did you run the python scripts? Under Linux or Windows?

If everything else fails, please zip your entire galagino directory containing the sketch and send it to me via email, so i can have a look and try to figure out which file differs and why.

C64fan, 2023-03-26:
Hi Till, Thank you, i have succeeded.Rebuild whole sketch and run the scripts 2nd time and it worked, the sampe_boom was indeed the culprit. I ordered a LiliGo T4 esp with 2.2 inch display for this project, it will be in a arcade casing, like Atari mini cabinet. On my YouTube channel I have some of my little projects, Mame Arcade cabinet with leds in the buttons.LiliGo Esp display with clock feed by ntp over wifi and ghost and pacman sprite running on background.Video and shorts ,  ***@***.*** I am interested in running 6502, z80 and 8088 games running natively in emulators, like, https://paoloseverini.wordpress.com/2016/02/13/galaga-an-arcade-machine-emulator-for-windows-and-html5/ and https://github.com/djyt/cannonball/wiki Thanks for this wonderfull project, a masterpiece. C64Fan

C64fan, 2023-03-31:
Hi Till,
Little progress, the project compliles but the display stays dark/gray.
Are you happy with sharing your TFT_eSPI User_Setup.h and User_Setup_Select.h config files ?
Greets,

harbaum, 2023-03-31:
I am not using any TFT library and thus no config files like that.

If you apply power and have the LED pin connected correctly then the ili9341 based display will light up bright white. If you have VCC (3.3V), GND and LED connected correctly then the screen should be white.

If that doesn't work and you are using a different setup than mine then you need to figure out how to configure Galagino for that specific board. You need to control the Backlight, drive the displays reset line correctly and finally setup the correct SPI pins.

I don't own one of the boards you use and can't easily help you with that.

C64fan, 2023-04-05:
Hi Till,
your answer was magic, many thanks.
Managed to resolve the pinning, all working ok.
#define TFT_CS  27, #define TFT_DC  32, #define TFT_RST 5, #define TFT_BL  4 for LilyGO TTGO T4 V1.3 ESP32.

harbaum, 2023-04-05:
Great. How about audio and the controls? How did you solve that?

I'll add your findings to the repository somehow so other TTGO T4 users can find tem.

C64fan, 2023-04-07:
Hi Till,
making the cabinet out of thin wood with smd audio amplifier, it will be only attraction mode in a pacman collection.
I tested the new attraction mode, splashscreen Pacman/Galaga  was ok, but the games give disturbed video on my Liligo T4.
The old project files function ok for the display.  (noticed the display rotation was changed in the new config file)
I will follow the progress.
Greets,

C64fan, 2023-04-07:
Pinned it down to the led routine, pin 16 on Liligo T4 needs to be disabled.
Master attrack only shows Pacman and splashscreen, not Galaga in next sequence.

C64fan, 2023-04-08:
Hi Till,
the define MASTER_ATTRACT_GAME_TIMEOUT  works fine for pacman, but not for galaga.
With define MASTER_ATTRACT_GAME_TIMEOUT  95000 , pacman is indeed 95 secondes but galaga only 37 seconds.
I will sent a picture when the mini cabinet is ready, will take some weeks to finish.

harbaum, 2023-04-08:
The game timeout part is not finished yet. This needs some major changes as the current solution of resetting the ESP32 in order to return to the main menu causes some crackling noise which I'd like to avoid if the system is supposed to be sitting quietly in master attract mode without bothering anyone.