harbaum/galagino #23: TypeError: object of type 'int' has no len()

jf1452, 2023-07-09:
PS C:\Software\Arcade\galagino-main\romconv> py ./cmapconv.py frogger_colormap ../roms/pr-91.6l ../galagino/frogger_cmap.h
Traceback (most recent call last):
  File "C:\Software\Arcade\galagino-main\romconv\cmapconv.py", line 222, in <module>
    palette = parse_palette(sys.argv[2])
              ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Software\Arcade\galagino-main\romconv\cmapconv.py", line 39, in parse_palette
    if len(c) == 3:
       ^^^^^^
TypeError: object of type 'int' has no len()

cmapconv.py returns an error when I try to run it. First I tried DigDug as it the new ROM, but also produces the same error when I try to convert an existing ROM.

I have tried py, python and python 3. Did I miss a setup step?

harbaum, 2023-07-09:
Thanks for reporting. This was a stupid bug introduced with the preparations for 1942.

closed 2023-07-09 by harbaum

jf1452, 2023-07-09:
Thank you Harbaum, that was a really quick update, thank you.

commit 83f323aa (2023-07-09): Fix #23