harbaum/galagino #24: redefinition of 'OpZ80_INL'

jf1452, 2023-07-09:
Apologies, I seem to be finding a few today.

When converting the galagino.ino into VSCode, then trying to compile, I get the message:

redefinition of 'OpZ80_INL'

Looking at the code, it's defined in emulation.h:

static inline byte OpZ80_INL(register word Addr) {
#ifndef SINGLE_MACHINE
  static const unsigned char *rom_table[][3] = {
    { NONE, NONE, NONE },
#define ROM_ENDL ,
#else
  static const unsigned char *rom_table[3] =
#define ROM_ENDL ;
#endif
#ifdef ENABLE_PACMAN 
    { pacman_rom, NONE, NONE } ROM_ENDL
#endif
#ifdef ENABLE_GALAGA
    { galaga_rom_cpu1, galaga_rom_cpu2, galaga_rom_cpu3 } ROM_ENDL
#endif
#ifdef ENABLE_DKONG
    { dkong_rom_cpu1, NONE, NONE } ROM_ENDL
#endif
#ifdef ENABLE_FROGGER
    { frogger_rom_cpu1, frogger_rom_cpu2, NONE } ROM_ENDL
#endif
#ifdef ENABLE_DIGDUG
    { digdug_rom_cpu1, digdug_rom_cpu2, digdug_rom_cpu3 } ROM_ENDL
#endif  
#ifndef SINGLE_MACHINE
  };
  return rom_table[machine][current_cpu][Addr];
#else 
  return rom_table[current_cpu][Addr];
#endif
}

and in z80.h:

static inline byte OpZ80_INL(register word Addr) {
#ifdef ENABLE_PACMAN
PACMAN_BEGIN
    return pacman_rom[Addr];
PACMAN_END
#endif
  
#ifdef ENABLE_GALAGA
GALAGA_BEGIN
    if(current_cpu == 1)       return galaga_rom_cpu2[Addr];
    else if(current_cpu == 2)  return galaga_rom_cpu3[Addr];
    return galaga_rom_cpu1[Addr];
GALAGA_END
#endif

#ifdef ENABLE_DKONG
DKONG_BEGIN
    return dkong_rom_cpu1[Addr];
DKONG_END
#endif

#ifdef ENABLE_FROGGER
FROGGER_BEGIN
   if(current_cpu == 0) return frogger_rom_cpu1[Addr];
   else                 return frogger_rom_cpu2[Addr];
FROGGER_END
#endif
}

If I delete the version in z80.h (because it seems the less complete version, all games run except DigDug which starts playing and digs halfway through the screen, then crashes.

harbaum, 2023-07-09:
Please re-run the z80 patch python script. If this shows up in two places then you probably updated your source tree from an older version but did not run these python scripts again.

I also encountered the digdug lockup during development. These were performance issues but they have been addressed.

Please make sure you have all files updated.

Did you try the using the Arduino IDE? If the problems only show up with Vscode, you may need to figure out the differences like e.g.compiler options or the like.

jf1452, 2023-07-09:
Thank you, running the scripts again fixed the duplicated definition... My bad!

DigDug still crashes just before Player 1 start. I installed Arduino 2.1.1 and can't get it to compile at all...

C:\Software\Arcade\galagino-main\galagino\video.cpp:52:1: sorry, unimplemented: non-trivial designated initializers not supported
 };
 ^
C:\Software\Arcade\galagino-main\galagino\video.cpp:52:1: sorry, unimplemented: non-trivial designated initializers not supported
C:\Software\Arcade\galagino-main\galagino\video.cpp: In constructor 'Video::Video()':
C:\Software\Arcade\galagino-main\galagino\video.cpp:127:43: error: 'SPI_DMA_CH_AUTO' was not declared in this scope
   spi_bus_initialize(VSPI_HOST, &bus_cfg, SPI_DMA_CH_AUTO);
                                           ^
In file included from c:\Users\jonfu\Documents\Arduino\libraries\FastLED\src/FastLED.h:75:0,
                 from C:\Software\Arcade\galagino-main\galagino\leds.h:8,
                 from C:\Software\Arcade\galagino-main\galagino\galagino.ino:14:
c:\Users\jonfu\Documents\Arduino\libraries\FastLED\src/fastspi.h:157:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^
C:\Software\Arcade\galagino-main\galagino\galagino.ino: In function 'void dkong_trigger_sound(char)':
C:\Software\Arcade\galagino-main\galagino\galagino.ino:488:23: error: no matching function for call to 'random()'
     char rnd = random() % 3;
                       ^
In file included from C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\8B6F6113B00CD58F3E8954C7FE93A26A\sketch\galagino.ino.cpp:1:0:
C:\Users\jonfu\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Arduino.h:121:6: note: candidate: long int random(long int, long int)
 long random(long, long);
      ^
C:\Users\jonfu\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Arduino.h:121:6: note:   candidate expects 2 arguments, 0 provided
In file included from C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\8B6F6113B00CD58F3E8954C7FE93A26A\sketch\galagino.ino.cpp:1:0:
C:\Users\jonfu\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Arduino.h:179:6: note: candidate: long int random(long int)
 long random(long);
      ^
C:\Users\jonfu\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Arduino.h:179:6: note:   candidate expects 1 argument, 0 provided
C:\Software\Arcade\galagino-main\galagino\galagino.ino: In function 'void audio_init()':
C:\Software\Arcade\galagino-main\galagino\galagino.ino:551:3: sorry, unimplemented: non-trivial designated initializers not supported
   };
   ^
C:\Software\Arcade\galagino-main\galagino\galagino.ino:551:3: sorry, unimplemented: non-trivial designated initializers not supported
C:\Software\Arcade\galagino-main\galagino\galagino.ino:551:3: sorry, unimplemented: non-trivial designated initializers not supported
C:\Software\Arcade\galagino-main\galagino\galagino.ino:551:3: sorry, unimplemented: non-trivial designated initializers not supported

exit status 1

Compilation error: 'SPI_DMA_CH_AUTO' was not declared in this scope

Should I be using the legacy Arduino?

harbaum, 2023-07-09:
I  just checked and everything still works fine for me. 

Are you sure you selected the correct esp32-dev board?

jf1452, 2023-07-10:
Thank you, your last message pointed me in the direction which allowed me to ask Google the right questions.

It turned out that Additional Boards Manager URLs picked up my config from when I last used it (years ago before I switched to VSCode). The Espressif links have been updated since then and now needs to point to https://espressif.github.io/arduino-esp32/package_esp32_index.json That done, it now compiles.

And the cherry on the top is that DigDug now runs. What the difference is, that will take a lot more investigation.