harbaum/galagino #25: Feature request

snakebyte69, 2023-07-15:
Versions that run on
- Agon Light
https://www.olimex.com/Products/Retro-Computers/AgonLight2/open-source-hardware
- ESP32-SBC-fabGL
https://www.olimex.com/Products/Retro-Computers/AgonLight2/open-source-hardware
- TTGO VGA
https://www.tindie.com/products/lilygo/lilygo-ttgo-vga-v14-controller/

harbaum, 2023-07-15:
These are all devices with VGA out. I don't think a ESP32 can handle VGA _and_ Galagino at the time.

snakebyte69, 2023-07-16:
Check out our Discord we have a channel for Galagino
https://discord.gg/yZXPAF56kX
I will send you each of those ESP32 SBCs

harbaum, 2023-07-16:
As I said, I am not sure the ESP32 can handle VGA and all the emulation stuff at the same time. So I am not even sure it's possible at all.

Thanks for the offer, but there's no need to send me anything. If I decide to give it a go, I'll just order some board that seems to fit the bill.

snakebyte69, 2023-07-17:
It wouldn't have to be full VGA resolution

snakebyte69, 2023-07-17:
https://github.com/rpsubc8/ESP32TinyPAC/tree/main/esp32/TinyPACttgovga32
This is a MAME PacMan port with VGA 6 bit colour