harbaum/galagino #26: Watchdog reboots DigDug and Galaga

KlausMu, 2023-07-26:
In "Dig Dug" and "Galaga" I get after some seconds:

`E (45312) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (45312) task_wdt:  - IDLE (CPU 0)
E (45312) task_wdt: Tasks currently running:
E (45312) task_wdt: CPU 0: emulation task
E (45312) task_wdt: CPU 1: IDLE
E (45312) task_wdt: Aborting.

abort() was called at PC 0x400f1021 on core 0
Backtrace:0x40083aed:0x3ffbe8ec |<-CORRUPTED
ELF file SHA256: 0000000000000000
Rebooting...
`
The other games work great.

harbaum, 2023-07-26:
Works here. Which version of the Arduino IDE do you use? Which version of the ESP32 board support? Are you using a ESP32 with or without PSRAM?

KlausMu, 2023-07-26:
I gave PlatformIO in Visual Studio Code a try. To me PlatformIO has a lot of advantages compared to the ArduinoIDE.
But as advised I also tried with the ArduinoIDE. Here all the games work (including DigDug and Galaga) - but unfortunately without sound.

Here the setup:

ArduinoIDE:
  IDE version 2.1.1
  ESP32 board support: esp32 [v 2.0.11]
  selected board: ESP32 dev board

PlatformIO:
  PlatformIO version 6.1.9
  PLATFORM: Espressif 32 (5.2.0) > Espressif ESP32 Dev Module
    framework-arduinoespressif32 @ 3.20005.220925 (2.0.5) 
  board = esp32dev

ESP32 hardware:
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash

harbaum, 2023-07-26:
It seems platformio produces slower code. Galagino really needs all performance for Galaga and Digdug. You might want to compare the compiler optimization settings both setups use.

With platformio you had sound? At least on the games that were working?

KlausMu, 2023-07-26:
Yes, with PlatformIO I had sound.

I will try to compare the compiler settings and see if I can finetune them. I will come back here and let you know if I was successfull.

KlausMu, 2023-07-27:
Ok, it really seems to be related to the ESP32 arduino version.
First I used v2.0.5 in PlatformIO. But there was an update available to version 2.0.9. With that it is running better, Galaga is running at least for lets say 10 seconds, but then it crashes again.
The arduino IDE is using v 2.0.11

I also tried some additional build flags seen in the Arduino IDE when having verbose output, but with no effect:
    -DHAVE_CONFIG_H
    -DUNITY_INCLUDE_CONFIG_H
    -DWITH_POSIX
    -D_GNU_SOURCE
    -DESP_PLATFORM
    -D_POSIX_READER_WRITER_LOCKS

I would use the Arduino IDE for compiling and flashing if I had sound there ...

harbaum, 2023-07-27:
I had the esp32 support version 2.0.6 installed in the Arduino IDE. I have now upgraded to 2.0.11 and indeed I only get what sounds like static noise and no game sound at all. Unfortunately, the 2.0.6 is not available, anymore. So I need to figure out what's the problem and try to fix that.

harbaum, 2023-07-27:
Edit: 2.0.10 is still available but has the same crackling noise ... annoying ...

KlausMu, 2023-07-27:
As a workaround, you can make 2.0.6 available again:
File - preferences - Additional boards managers URLs - add
https://espressif.github.io/arduino-esp32/package_esp32_index.json

Then
Tools - Board - Boards Manager
Now you can select version 2.0.6 for "esp32 by Espressif Systems"

No I have sound and Galaga running.

Good luck for finding the issue and thanks a lot for this amazing project.

harbaum, 2023-07-29:
Please use the Arduino IDE

closed 2023-07-29 by harbaum

rlueckl, 2023-10-20:
Had the same crashing issue with VS Code **and** Arduino IDE with 2.0.11.

Downgrading Arduino IDE esp32 Espressif to 2.0.6 fixed it for me also.