harbaum/galagino #27: Crackling sound

harbaum, 2023-07-27:
While sound was working fine with ESP board support 2.0.6 it only gives a crackling noise in the 2.0.10 and the latest 2.0.11

harbaum, 2023-07-27:
Works up to 2.0.9 and is broken since 2.0.10

The following sketch gives a clean rectangular signal up to 2.0.9 and the crackling noise on later versions.

```
#include "driver/i2s.h"

unsigned short snd_buffer[64];

void setup() {
  // put your setup code here, to run once:

  static const i2s_config_t i2s_config = {
    .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_TX | I2S_MODE_DAC_BUILT_IN),
    .sample_rate = 24000,
…  printf("BytesOut: %d\n", bytesOut);
}

```

harbaum, 2023-07-27:
See https://github.com/espressif/arduino-esp32/issues/8467

harbaum, 2023-07-29:
Please try the latest commit. The donkey kong sounds are the most critical ones and may be affected most by the implemented work around.

Still unclear what's causing the apll to generate the crackling noise.

bjotos, 2023-08-01:
I started with this project after reading about it in the latest Make and ran in exact the issue with crackling noise from the speakers w/o this fix (cloned repo last week). And this commit solved it.

But I was getting this noise also in game select screen where it should be silent, which makes me wonder if that's because I'm using single-wire SC8002B, while PAM8302A uses differential audio signal, or some other difference with these solutions (different amps, different filters, ...)

P.S.: Thanks for the great work, love it ;)

harbaum, 2023-08-02:
The crackling sound even happens when there's silence at a signal level != zero. So yes, it was expected to happen in the menu as well.

closed 2023-08-02 by harbaum