harbaum/galagino #28: upload is not executed

Moritz24503, 2023-07-28:
Hey, when I want to upload the code via the Arduino IDE, the following error message always comes up. What can I do about it?`

`/Users/benutzer/Desktop/galagino-main3/galagino/video.cpp:52:1: sorry, nicht implementiert: nicht-triviale designierte Initialisierer werden nicht unterstützt }; ^ /Users/benutzer/Desktop/galagino-main3/galagino/video.cpp:52:1: sorry, nicht implementiert: nicht-triviale designierte Initialisierer werden nicht unterstützt /Users/benutzer/Desktop/galagino-main3/galagino/video.cpp: Im Konstruktor 'Video::Video()': video.cpp:131:22: Fehler: 'SPI1_HOST' wurde in diesem Bereich nicht deklariert spi_bus_initialize(SPI1_HOST, &bus_cfg, SPI_DMA_CH_AUTO); ^ video.cpp:131:43: Fehler: 'SPI_DMA_CH_AUTO' wurde in diesem Bereich nicht deklariert spi_bus_initialize(SPI1_HOST, &bus_cfg, SPI_DMA_CH_AUTO); ^ In der Datei enthalten von /Users/benutzer/Documents/Arduino/libraries/FastLED/src/FastLED.h:75:0, von /Users/benutzer/Desktop/galagino-main3/galagino/leds.h:8, von /Users/benutzer/Desktop/galagino-main3/galagino/leds.cpp:7: /Users/benutzer/Documents/Arduino/libraries/FastLED/src/fastspi.h:157:23: Hinweis: #pragma-Nachricht: Keine Hardware-SPI-Pins definiert. Alle SPI-Zugriffe werden standardmäßig auf Bitbang-Ausgaben gesetzt # pragma-Nachricht "Keine Hardware-SPI-Pins definiert. Alle SPI-Zugriffe werden standardmäßig auf bitbanged-Ausgaben gesetzt" ^ In der Datei enthalten von /Users/benutzer/Documents/Arduino/libraries/FastLED/src/FastLED.h:75:0, von /Users/benutzer/Desktop/galagino-main3/galagino/leds.h:8, von /Users/benutzer/Desktop/galagino-main3/galagino/galagino.ino:14: /Users/benutzer/Documents/Arduino/libraries/FastLED/src/fastspi.h:157:23: Hinweis: #pragma-Nachricht: Keine Hardware-SPI-Pins definiert. Alle SPI-Zugriffe werden standardmäßig auf Bitbang-Ausgaben gesetzt # pragma-Nachricht "Keine Hardware-SPI-Pins definiert. Alle SPI-Zugriffe werden standardmäßig auf bitbanged-Ausgaben gesetzt" ^ /Users/benutzer/Desktop/galagino-main3/galagino/galagino.ino: In der Funktion 'void dkong_trigger_sound(char)': galagino:555:23: Fehler: keine Matching-Funktion für den Aufruf von 'random()' char rnd = random() % 3; ^ In der Datei enthalten von /var/folders/6c/04z3f_2j7kvd7nx0xxnjn3g40000gn/T/arduino_build_284766/sketch/galagino.ino.cpp:1:0: /Users/benutzer/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/cores/esp32/Arduino.h:121:6: Hinweis: Kandidat: long int random(long int, long int) lang zufällig (lang, lang); ^ /Users/benutzer/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/cores/esp32/Arduino.h:121:6: Hinweis: Der Kandidat erwartet 2 Argumente, 0 zur Verfügung gestellt In der Datei enthalten von /var/folders/6c/04z3f_2j7kvd7nx0xxnjn3g40000gn/T/arduino_build_284766/sketch/galagino.ino.cpp:1:0: /Users/benutzer/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/cores/esp32/Arduino.h:179:6: Hinweis: Kandidat: long int random(long int) lang zufällig (lang); ^ /Users/benutzer/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/cores/esp32/Arduino.h:179:6: Hinweis: Kandidat erwartet 1 Argument, 0 zur Verfügung gestellt /Users/benutzer/Desktop/galagino-main3/galagino/galagino.ino: In der Funktion 'void audio_init()': /Users/benutzer/Desktop/galagino-main3/galagino/galagino.ino:618:3: sorry, nicht implementiert: nicht-triviale designierte Initialisierer nicht unterstützt }; ^ /Users/benutzer/Desktop/galagino-main3/galagino/galagino.ino:618:3: sorry, nicht implementiert: nicht-triviale designierte Initialisierer nicht unterstützt /Users/benutzer/Desktop/galagino-main3/galagino/galagino.ino:618:3: sorry, nicht implementiert: nicht-triviale designierte Initialisierer nicht unterstützt /Users/benutzer/Desktop/galagino-main3/galagino/galagino.ino:618:3: sorry, nicht implementiert: nicht-triviale designierte Initialisierer nicht unterstützt Bibliothek FastLED in Version 3.6.0 im Ordner: /Users/benutzer/Documents/Arduino/libraries/FastLED wird verwendet Ausstiegsstatus 1 'SPI1_HOST' wurde in diesem Bereich nicht deklariert`

harbaum, 2023-07-29:
What version of the Arduino IDE are you using? Which version of the ESP32 board package? What board type did you select?

Moritz24503, 2023-07-31:
Arduino Ide: 1.8.19
ESP32 Package: 1.0.6
Board Type: ESP32 Wrover Module

harbaum, 2023-07-31:
Please use esp32 bosrd support 2.0.9

And don't enable the PSRAM of the wrover.

harbaum, 2023-08-14:
Does it work now?

Moritz24503, 2023-08-15:
Yes, thanks.

closed 2023-08-15 by Moritz24503