harbaum/galagino #31: Volume Settings

dirtyharryiv, 2023-08-11:
Hi,

is it possible to change the volume per software or do i have to use a potentiometer?
Great Project btw!

greets
Harry

harbaum, 2023-08-11:
There is no volume adjustment in software. The DAC has only 8 bit resolution and I am not sure how much an additional volume control would degrade the sound even further.

I was actually thinking about some kind of mute function to annoy the colleagues a little less ... but that has not been implemented, yet.

bjotos, 2023-08-12:
Both PAM8302 and SC8002B have a volume pot, you'd just need to make it accessible from outside of the cabinet.
At least with my SC8002B I'm able to completely mute the audio.

dirtyharryiv, 2023-08-14:
The volume pod on the PAM8302 is so small i didn't see it. Thanks for the info! Now i just have to figure out how to adjust the volume on that tiny thing...