harbaum/galagino #35: Missing i2s libs in the sources

purehunter1, 2023-12-11:
When trying to compile the galagino.ino I encounter an error message, that the file "driver/i2s.h" is missing. Due to the call
#include "driver/i2s.h"
it has to be part of the sourcefiles, but it isn't.

I tried to collect this and other missing files from espressif and place it in the respective subdirectory of the sourcefiles but this is a never-ending story.
Is there a simpler and safe soution in place?

Regards purehunter

purehunter1, 2023-12-12:
It seems, it has to do with the actual ESP32 Board Manager 2.0.11. After downgrading to Rev. 2.0.10 and modification of the #include statement to
`#include <driver/i2s.h>`
I got the compilation completed.
But I'm waiting for the display, so I cannot test it right now.

I don't want to miss to say: Absolutely great project! I'm really happy that I found it. Thanks for all the work!

Regards purehunter

harbaum, 2023-12-26:
If you figure out how to use this with the newer versions, then please feel free to open a PR.

purehunter1, 2024-01-04:
I just tried to reproduce this problem, but after upgrading the board-manager to 2.0.11 again, it still worked.
But only with the modification mentioned above: replace `#include "driver/i2s.h"` by `#include <driver/i2s.h>`

As I got some questionable error-messages and warnings (only) at the first compilation after upgrading, I think it is a kind of instability of the IDE.

closed 2024-01-04 by purehunter1

harbaum, 2024-01-04:
Thanks. Please open a PR for this.