harbaum/galagino #36: Gerber Files for cabling

AxelAhlborn, 2023-12-22:
Hello Mr. Harbaum, in your YouTube video https://www.youtube.com/watch?v=s7kZmB4fsLA you show circuit boards for the cabling from 42:30 onwards. I'm very interested in this. Prototype on breadboard is already running, joystick is also ready. Housing parts created with the 3D printer. The wiring is a horror because I'm currently unable to keep my hand as steady as I should for soldering. Can you please include the Gerber files in the project. Thank you ! Many greetings from the Wolfenbüttel district. Axel Ahlborn

harbaum, 2023-12-22:
Indeed, I never uploaded these. I have to admit that I also never tested them as I had enough devices built the ordinary way. Will upload them asap ...

AxelAhlborn, 2023-12-23:
Hello Mr. Harbaum, I would be happy to support you with the tests. Even if I have to place further orders with JLCPCB, it won't be a big hassle.

AxelAhlborn, 2023-12-23:
axel.ahlborn@googlemail.com

harbaum, 2023-12-26:
Added: https://github.com/harbaum/galagino/tree/main/hardware/pcbs

closed 2023-12-26 by harbaum