harbaum/galagino #38: Feature request: more ROM options

giobbino, 2024-01-10:
Absolutely amazing! Awesome! I was looking for some fun projects with ESP32, and after some failures with other projects, I found this one, that works like a charm with great and very complete instructions. Instead of a cabinet, I 3D printed a console for it (I will release it on thingiverse as soon as I will reach the final version (no issues).

I was wondering: can you release other ROM packs with different games of the era? I wish make other consoles using the same hardware; actually I looked at the source code, but building a new ROM pack is beyond my skills, unfortunately.

I would love to see Gyruss in this pack, that was quite famous (at least in some part of Europe) and has an amazing music.
Other classic ROMs of the era with vertical display (IIRC), just as suggestion/example, could be Space Invaders, Phoenix, Xevious, Time Pilot, Quix, Q-Bert, Bomb Jack...

harbaum, 2024-01-11:
Great that you are having fun with it.

There's a general issue with adding more games. Currently, there's one CPU emulation code which decides for every instruction for which machine this is to be emulated. This gets slightly slower with every machine added and affects emulation speed of the existing machines. But this can sure be optimized with some effort.

Regarding the games you are suggesting: 
- Gyruss has four CPUs of three different types and five audio chips. That's simply beyond what the ESP32 or at least my solution for it is able to handle
- Space invaders is IMHO to close to Galaga. So is Phoenix. Both are nice but "yet another old space shooter" doesn't motivate me enough to spend the time
- xevious may work and even runs on the same hardware as 1942. So that's an option
- Time Pilot may also work. But is yet another shooter like Galaga and 1942
- Quix looks "different" which is fine. But it's based on the 6809 CPU which I don't have an emulation for
- Q-Bert also is nice and also has a CPU which I don't emulate (yet): a 8086 CPU
- Bomb Jack seems to fit the bill quite well. Two Z80 CPUs and three AY sound chips are e.g. what 1942 also uses

To summarize: I have no immediate plans to work on this. But if I had to choose one game it would probably be Bomb Jack. It differs enough from the games already implemented to add something. And its hardware seems to be pretty close to what Galagino already supports. Still it's some days work to implement.

This is a nice description how Bomb Jack works: https://floooh.github.io/2018/10/06/bombjack.html

giobbino, 2024-01-11:
Hi, thanks for your prompt answer! 

too bad Gyruss isn't an option, but okay, I can play it on my PC, so not a big issue  :-)

Of course all the games I mentioned were just suggestions/examples, not a real requirement; I just cited the ones that was really cool in the '80s, based on my memory. 

A note about Space Invaders: I friendly disagree with you; yes it's another space shooter and it's close to Galaga, but it's an icon, so it would be really nice because of its historical relevance. Just a personal opinion, of course. 

Time Pilot is actually another shoot game, but IMO it's quite different from the others because you don't move your ship from the center of the screen, so the player experience is really different and not comparable with 1942, I believe.

Other titles I didn't mentioned (yet) because I was not even sure you could be some way interested to spend more time on this project (of course I'm strongly hoping you will do LOL). I would say "Tetris" but I'm afraid it would be too much for the ESP32 too. 

Maybe Burger Time? Tapper?

Bomb Jack would be awesome! I'm reading the page you cited, absolutely amazing! I've never think about what was hard for the programmers of that era to fit some incredible games in a tight hardware!

gilmod714, 2024-09-28:
I'm surprised nobody mentioned "Horizontal space shooters" like Defender and Gradius ! .. how about adding this variety to the mix ;)