harbaum/galagino #39: Constant buzzing from speaker on GPIO26 (CYD). Can an external I2S amp be used?

pinter75, 2024-01-24:
HI,

Steller work on this, it's faultless!! 

Except for the sound on my CYD display. It has an amp on pin 26, but all I hear is a buzzing sound?

To be honest I have not really ever had much luck with the sound on these boards.

So can an external I2S amp (MAX98357) be used, probably overkill for the sounds generated, but a least I will hear something less annoying than a buzzing!

:-)

pinter75, 2024-01-25:
Seems I have a broken amp IC.

closed 2024-01-25 by pinter75

not-surt, 2024-01-25:
There's a known design defect with the CYD amp: https://github.com/witnessmenow/ESP32-Cheap-Yellow-Display/blob/main/Mods/README.md#audio-amp-gain-mod

harbaum, 2024-01-25:
It should be possible to use an external i2s amp with minimal changes. But I never tried that.