harbaum/galagino #43: Higher quality reference image?

GMDSulky, 2024-02-07:
Hey there! I just found this project and it's REALLY cool! I can't wait to make little tiny arcade machines in my own home!

I was just wondering if you had a higher quality image of the reference breadboard, I'm having trouble figuring out what wires go where as I think the pins on my ESP32 are laid out differently compared to the example. Here is the ESP32 I'm using for reference.

https://www.aliexpress.us/item/3256805517875317.html?spm=a2g0o.order_list.order_list_main.17.7c8b1802ZVvMCu&gatewayAdapt=glo2usa

Thanks in advance!

- Sulky

harbaum, 2024-02-07:
There's a PDF link right below the image. This will give you a much higher resolution.

GMDSulky, 2024-02-07:
Ah. Didn't see the hyperlink. My bad. Thanks.

closed 2024-02-07 by GMDSulky