harbaum/galagino #49: CYD white background fix

DynaMight1124, 2024-07-04:
Hi

Firstly, thanks for the amazing project. I have recently built a small arcade with this code and its amazing!

I also have a CYD2USB with a ST7789 display, however when I flashed I had a white background rather than black. I know from other projects I've used on this device that it can do that with this display.

I have managed to correct the issue, I'm hoping my below fixes can be added and hoping I dont need to submit a PR because I have no idea how to do that :)

So anyway my fixes involve both config.h and video.cpp.

In config.h I added this line to line 80:
// #define TFT_INV_OFF // some CYD screens need inversion off if you have a white background rather than black
Normal behaviour should probably be commented so it doesn't affect other ST7789 users.

In video.cpp I deleted line 104 and added:
#ifdef TFT_INV_OFF
  0x20, 0,                          // INV OFF
#else
  0x21, 0,                          // INV ON
#endif

I tested the line in config.h commented out and in and it had the correct effects on the display indicating the code is working e.g. white and black backgrounds.

Thanks

AnDyC420, 2024-12-05:
Thank you for posting this.

nickhastreiter, 2025-02-25:
Thank you for posting this!

nickhastreiter, 2025-02-25:
Thank you for posting this