harbaum/galagino #56: Help with a custom controller implementation on CYD.

aedile, 2024-12-01:
Greetings,

I have things working on a CYD, but I don't want to use a Wii nunchuck.  CYD doesn't have adequate GPIO pins for a proper controller, so I've worked on making my own i2c device using an Seeeduino Xiao.  I tested this i2c device using an ESP32 and it seems to work just fine when using the default i2c pins on the esp32.  Here is the slave sketch:

```
#include <Arduino.h>
#include <Wire.h>

// Define Slave I2C Address
#define SLAVE_ADDR 9

#define XPIN  0 // Arduino pin connected to VRX pin
#define YPIN  1 // Arduino pin connected to VRY pin
#define ZPIN  2  // Arduino pin connected to VRZ pin

#define APIN 3 // Arduino pin connected to button A pin
#define BPIN 6 // Arduino pin connected to button A pin

// Input values
int joystick_calibration_x = -3;
int joystick_calibration_y = 6;
int xValue = 0; // To store value of the X axis
int yValue = 0; // To store value of the Y axis
int xPinValue = 0;
int yPinValue = 0;
bool zButton = 0; // To store value of the Z button
bool aButton = 0; // To store value of the A button
bool bButton = 0; // To store value of the B button

// Banger
int bcount = 0;

// I2C Communication
void requestEvent() {
  // Define a byte to hold data
  byte bval = 0;
  
  // Cycle through data
  // First response is always 255 to mark beginning
  switch (bcount) {
    case 0:
      bval = 255;
      break;
    case 1:
      bval = xValue;
      break;
    case 2:
      bval = yValue;
      break;
    // Note: Buttons read high normally, so we invert them  
    case 3:
      bval = zButton;
      break;
    case 4:
      bval = aButton;
      break;
    case 5:
      bval = bButton;
      break;
  }
  
  // Send response back to Master
  Wire.write(bval);
  Serial.printf("Sent: %d\n", bval);
  // Increment byte counter
  bcount++;
  if (bcount > 5) bcount = 0; 
}

void setup() {
[... 192 lines cut ...]
Byte is currently:1
Waiting to end read
Byte is currently:1073429516
Waiting to end read
Byte is currently:127
Waiting to end read
Byte is currently:127
Waiting to end read
Byte is currently:1
Waiting to end read
Byte is currently:1
Waiting to end read
```
This is really strange to me, but I am new to i2c, Arduino IDE (platformio guy), and Galagino.  I have tried the following things:
1. Using TwoWire instead of wire, 0, 1, and 9 as indices for the constructor (9 just for funsies, it compiled and worked the same as 0, and 1 which behaved the same as just plain "Wire".
2. Swapping both the code and the wiring of the SDA and SCL (and NO I didn't pull a two doctors BOTH reversing the polarity when making that test). I got hung up at a different area when I did this and confirmed I am wired correctly for the CYD with my current setup.
3. Initializing the pins with pullup resistors (this seems prudent as I don't know if the Wire library will initialize the pins appropriately, but may not be necessary, I've tried it both ways and left it in for now). I realize this may be part of the issue since I have extended the wires by necessity at this point due to breadboarding needs. I will try adding pullups inline if nothing else works by tomorrow (not in my workshop rn so no resistors on hand).
4. Adding a bunch of serial debugging messages - this helped narrow down where I am going wrong.
5. Reinitializing the wire each time - this adds way too much latency and also doesn't work.

Things I haven't tried:
1. Just setting it to say if the byte is equal to "1073429516" assume it is the start byte. I don't want to do this because I don't understand why I'm getting that number and it seems like cheating. I may try this later just for funsies, but it's NOT a solution.
2. Adding in the pullup resistors. I will likely try this tomorrow.

Any suggestions are welcome.  Thanks for the great software!  So cool!!

harbaum, 2024-12-03:
There are a few things that don't make much sense. The value printed is way too big for a single byte. This often indicates that signed and unsiged are messed up, but in this case the value is 0x3ffb3c0c which actually isn't negative as a 32 bit value either. You really need to figure out what this byte type is. sizeof(byte) may give some hints.

Another thing I noticed is that bval is neither initialized nor set later in case of a timeout. So in that case readi2c would return random stuff from the stack in case of a timeout,

This all isn't Arduino or galagino specific but rather just plain C. You simply need to debug this further. What exact size is a "byte" here? Is it signed or unsigned? How is this expanded when doing a printf? etc etc ...

aedile, 2024-12-15:
@harbaum  - Thank you.  I am a programming generalist and you could inscribe what I know about C and C++, specifically, on a grain of rice with a blunt crayon.  That said, I think this is definitely something to consider.  Appreciate your feedback.  Will persist with debugging and hopefully post working code here soon!

aedile, 2024-12-15:
I actually accidentally posted my last update to a completely different issue board like a dork (bluepad32 board) anyways, I have changed tracks, and tried to move to a bluetooth controller implementation. I thought I'd take a crack at getting the bluepad32 library working and I've mostly succeeded.  
https://github.com/ricardoquesada/bluepad32
For now I've just made a custom implementation of Nunchuck.h.  Here is that file:

```
#ifndef _NUNCHUCK_H_
#define _NUNCHUCK_H_

#include <Bluepad32.h>

ControllerPtr gamepad = nullptr;

void onConnectedController(ControllerPtr gp) {
    gamepad = gp;
    Serial.println("Gamepad connected");
}

void onDisconnectedController(ControllerPtr gp) {
    if (gamepad == gp) {
        gamepad = nullptr;
        Serial.println("Gamepad disconnected");
    }
}

void nunchuckSetup() {
    delay(3000);
    BP32.setup(&onConnectedController, &onDisconnectedController);
    Serial.println("Bluepad32 setup completed");
}

unsigned char getNunchuckInput() {
  bool dataUpdated = BP32.update();
  if(gamepad && dataUpdated){
    Serial.printf("idx=%d, dpad: 0x%02x, buttons: 0x%04x, axis L: %4d, %4d, A: %d, B: %d, X: %d, Y: %d\n",
        gamepad->index(),        // Controller Index
        gamepad->dpad(),         // D-pad
        gamepad->buttons(),      // bitmask of pressed buttons
        gamepad->axisX(),        // (-511 - 512) left X Axis
        gamepad->axisY(),        // (-511 - 512) left Y axis
        gamepad->a(),            // A button
        gamepad->b(),            // B button
        gamepad->x(),            // X button
        gamepad->y()             // Y button
    );
    int joyY = gamepad->axisY();
    int joyX = gamepad->axisX();

    return ((joyX < -127) ? BUTTON_LEFT : 0) | //Move Left
            ((joyX >  127) ? BUTTON_RIGHT : 0) | //Move Right
            ((joyY < -127) ? BUTTON_UP : 0) | //Move Up
            ((joyY >  127) ? BUTTON_DOWN : 0) | //Move Down
            (gamepad->a() ? BUTTON_FIRE : 0) |
            (gamepad->b() ? BUTTON_EXTRA : 0) ;
  }else{
    Serial.println("No gamepad");
    return 0;
  }
  vTaskDelay(1);
}

#endif //_NUNCHUCK_H_
```

The cool thing is this is MOSTLY working.  When the library is compiled with the Bluepad32 version of ESP32 Dev Module, everything compiles, the controller is detected.  The buttons work and Up and Down work in the menu for picking which game you want to play.  The problem I'm having right now, and I'm just completely stuck on, is that, using the serial prints, I know I'm getting left and right from the x and y axes (-512 for left/up, 512 for rt/dn on press with it centered at maybe 3 on x axis and -3 on y axis).  I've even tried some debugging to verify that when dpad is pushed, that `getNunchuckInput` is returning a non-zero value different from normal with distinct values for left, right, up and down.  It seems to work fine in the menu, just not in any of the games.  Not sure what I am doing wrong here, unfortunately.  I feel like it's something to do with the return object.  Here is what the original Nunchuck.h returns:

```
return ((joyX < 127 - NUNCHUCK_MOVE_THRESHOLD) ? BUTTON_LEFT : 0) | //Move Left
           ((joyX > 127 + NUNCHUCK_MOVE_THRESHOLD) ? BUTTON_RIGHT : 0) | //Move Right
           ((joyY > 127 + NUNCHUCK_MOVE_THRESHOLD) ? BUTTON_UP : 0) | //Move Up
           ((joyY < 127 - NUNCHUCK_MOVE_THRESHOLD) ? BUTTON_DOWN : 0) | //Move Down
           (nchuk.buttonZ() ? BUTTON_FIRE : 0) |
           (nchuk.buttonC() ? BUTTON_EXTRA : 0) ;
  }
```
I feel like what I'm returning is not markedly different (just different expected values from the joystick).

Example serial output dead stick:
```
idx=0, dpad: 0x00, buttons: 0x0000, axis L:    3,   -3, A: 0, B: 0, X: 0, Y: 0
idx=0, dpad: 0x00, buttons: 0x0000, axis L:    3,   -3, A: 0, B: 0, X: 0, Y: 0

```

Example serial output left stick:
```
idx=0, dpad: 0x00, buttons: 0x0000, axis L: -512,   -3, A: 0, B: 0, X: 0, Y: 0
idx=0, dpad: 0x00, buttons: 0x0000, axis L: -512,   -3, A: 0, B: 0, X: 0, Y: 0
```

My only other guess is that MAYBE something about the bluepad32 library itself is causing issues detecting input.  Not 100% sure if that's the case, but stranger things have happened.  Anyways, delving into the C code where control is parsed is not something I'm skilled enough to do with any sort of speed, so at this point, while I am working my way through it, thought I'd ask again since you were helpful with the first version.  I'm incredibly close to things working 100% right now.  Any suggestions are welcome.  For reference, I'm not at all a C or C++ guy but I am fluent in psuedocode and many other languages, as well as googling and R-ing TFM.  

If I can get this updated, it's a few simple steps to a working PR.  Bluepad32 works with many controllers, you've probably got one laying around.  If you'd like to try this out, follow the instructions for the bluepad32 installation in arduino ide here: https://github.com/ricardoquesada/bluepad32 , compile galagino for the CYD board under bluepad32 esp32 instead of esp32 and drop in the above file for Nunchuck.h, it compiles and lets you select a ROM in the menu.  Works without the controller connected as well in "demo mode" where it selects a master on it's own.  I've got this last little issue to clear up is all.  Cheers!

aedile, 2024-12-15:
Sorry, for the above, button presses are detected in the menu and the game, but not directions.