harbaum/galagino #62: Compilation error: implicit declaration of function 'millis' [-Wimplicit-function-declaration]

fedekrum, 2025-04-07:
Hi
I am on a Mac, using Arduino IDE Version: 2.3.5
Date: 2025-04-02T13:09:23.254Z (4 days ago)
CLI Version: 1.2.0

I am using board "ESP32-2432S028R CYD" in the board menu and have one of those plugged into the USB connector.

I can run other sketches successfully.

This is the error with Galagino.

/Volumes/Development/galagino/galagino/emulation.c: In function 'prepare_emulation':
/Volumes/Development/galagino/galagino/emulation.c:560:28: error: implicit declaration of function 'millis' [-Wimplicit-function-declaration]
  560 |   master_attract_timeout = millis();
      |                            ^~~~~~
/Volumes/Development/galagino/galagino/emulation.c: In function 'emulate_frame':
/Volumes/Development/galagino/galagino/emulation.c:573:23: error: implicit declaration of function 'micros' [-Wimplicit-function-declaration]
  573 |   unsigned long sof = micros();
      |                       ^~~~~~
/Volumes/Development/galagino/galagino/emulation.c:623:37: error: implicit declaration of function 'esp_random'; did you mean 'srandom'? [-Wimplicit-function-declaration]
  623 |       machine = 1 + ((unsigned long)esp_random()) % MACHINES;
      |                                     ^~~~~~~~~~
      |                                     srandom
/Volumes/Development/galagino/galagino/emulation.c:689:5: error: implicit declaration of function 'ulTaskNotifyTake' [-Wimplicit-function-declaration]
  689 |     ulTaskNotifyTake(1, 0xffffffffUL);
      |     ^~~~~~~~~~~~~~~~
/Volumes/Development/galagino/galagino/emulation.c:692:5: error: implicit declaration of function 'vTaskDelay' [-Wimplicit-function-declaration]
  692 |     vTaskDelay(1);
      |     ^~~~~~~~~~
exit status 1

Compilation error: implicit declaration of function 'millis' [-Wimplicit-function-declaration]

elandivar, 2025-04-13:
Include the line:

#include <Arduino.h>

At the top of emulation.c file

elandivar, 2025-04-13:
Downgrade the ESP32 board to 2.0.17. At least, it worked for me.

fedekrum, 2025-04-19:
> Or... downgrade the ESP32 board to 2.0.17. At least, it worked for me.

It compiled with 3.0.7 and 3.1.3 too.
The problem is 3.2.x

harbaum, 2025-04-19:
Thanks for the info. Even if it compiles and runs there may still be timing/performance issues. 

Such problems usually manifested in the dig-dug title animation. If that runs then the overall timing and performance should be fine.

Please consider submitting a PR should you fix things for later IDF versions.

fedekrum, 2025-04-19:
> Please consider submitting a PR should you fix things for later IDF versions.

Sorry, but it is probably out of my capabilities.